openair-collective/openair-epiphyte #15: ENGR: Improve/reduce resistance of airflow through sorbent panel

KCollins, 2024-06-04:
Related to Issue #14 .

With our current sorbent panel, the sorbent is tightly packed in to maximize quantity and therefore absorption capacity. But this causes a lot of resistance for airflow, which also has an energy cost. We need to rethink how we can get the same amount or more sorbent ih the device, while also allowing for less resistance.

lingxkong, 2024-06-16:
This is an idea relating to the intrinsic properties of how the sorbents settle based on the way the device is build. The chamber is a rectangular prism slab with the two long sides equal in length, similar to a square drink coaster. The slab sits on one of its thin sides, causing the sorbent to settle down toward the bottom thin side through the force of gravity. This idea has been documented in the[ #epiphyte channel](https://discord.com/channels/656929401933529088/1059420084583546890/1249622156409831457) on the [Open Air discord](https://openaircollective.com/join/).

**Problem it’s supposed to solve:** If the sorbent chamber is not entirely filled, any gaps will cause the air passing through to take the path of least resistance, particularly if the sorbent is powder form rather than pellets. We would like to force a certain amount of air through portions of the sorbent window, particularly at the bottom.

**Description of idea/proposed solution:** It’s a tunnel interfacing the sorbent window with the fan (there is only one fan at the output which pulls air through using negative pressure). The key with this tunnel is that it’s divided up into portions—maybe 4, 9, or 16 divisions so that each division gets its own window of sorbent to work on, i.e., pull air through.

(please excuse the hand-drawn)
[image]

**Materials:** There is a supplier called TAP Plastics that I have access to. I can’t see why suppliers similar to McMaster would also have plexiglass or polycarbonate pipes, and I believe I have seen square ones. Mounting is a separate issue and I haven’t optimized it in my head—maybe a supporting structure from the bottom, or some L brackets to mount.

**Potential technical difficulties and their workarounds:** I think this idea will by no means go smoothly. I am already worried about implementation

- may need sealing between the sorbent window and the tunnel. May need to create a custom part to do this. This gets mechanically complicated quickly. Not to mention the fan side also.

- If the sorbent is powder, there may be no way to stir up the sorbent and just applying negative pressure will just cause clumping. Is there a way to reverse the fan once in a while? Moving to positive pressure and turning the whole device so that the inlet is at the bottom and outlet is at the top may also help. This is, of course, not easy.

See also [this post](https://github.com/openair-collective/openair-epiphyte/issues/11#issuecomment-2171931588) in issue #11 for another angle for air flow-through.

mxbifurcations, 2024-07-26:
[image]
A search on "fluidized bed with vortex" shows a variety of designs made to move solid grains in a gas flow. A wood chips / dust collector works the same way. If hot air is blown the chamber could also perform the desorption ?