openair-collective/openair-epiphyte #28: Find out how to make a wollastonite slurry/best mixing ratio

Charles-DCole, 2024-11-14:
Perform multiple tests with different mixtures of slurry to find the best amount.

kavyaashah, 2024-11-21:
We started experiments in the workshop. We're testing 1:1, 2:1, 3:4, and 4:3 ratios of wollastonite to water and checking it to observe separation and change in pH levels.

evpat1299, 2024-12-06:
I came across a paper recently that examined water:wollastonite ratios of 2-10:1, and another paper that just used a ratio of 10:1 for their experiments based on the findings of the first paper I mentioned. Might be worth it to examine higher water ratios.

https://www.sciencedirect.com/science/article/pii/S0009250906001059
This is the study that examined the liquid/solid ratios, and while there was a slight increase in CO2 conversion with increasing L/S ratio, they did mention they weren't sure if it was significant.