openair-collective/openair-epiphyte #35: Make Base (KOH) Baths to surface clean monoliths

penguins4u2, 2024-11-21:
Prepare large scale solution of KOH base bath for surface cleaning monoliths 

(need to determine what concentration of KOH to make for the baths) 

Items needed: 

- 3 x Buckets (at least ~1 gallon size) 
- 10 Gallons of DI water 
- KOH pellets

penguins4u2, 2024-11-21:
Before KOH bath submerging , need to pre-clean monoliths with ethanol & dry under vacuum. 

- submerge in ethanol for ~ 5mins, repeatedly removing and resubmerging 
- dry under vacuum at 100C for ~1hr 

once dried, the monoliths are ready for KOH treatment.

Aaron-T-A, 2025-02-07:
Clarity Tech Monolith has undergone KOH bath (I think. Check me on this Will)
Base bath done with a molarity around 0.7, but more basic bath could be more beneficial.